gianluca-nitti/printserver-esp8266 #1: aggiungere ref a documentazione Centronics (parallela)

atrent, 2018-01-09:


gianluca-nitti, 2018-01-10:
Il documento di specifica dello standard è questo: http://ieeexplore.ieee.org/document/469123/

Ho scoperto che aprendo ieeexplore.ieee.org da rete eduroam c'è il link diretto per scaricare il PDF mentre quando avevo provato con una connessione "normale" (di casa mia) mi proponeva di acquistare (se ricordo bene) il singolo PDF o una membership IEEE, quindi lo avevo scaricato usando il mirror unimi (che mi chiedeva il login) a questo link http://ieeexplore.ieee.org.pros.lib.unimi.it/document/469123/

A breve aggiungo un readme al repository con questo e vari altri link utili.

atrent, 2018-01-10:
Unimi ha abbonamento ieee, acm e tante altre libraries :)

atrent, 2018-01-10:
Cmq bastava un link a wikipedia ;)