gianluca-nitti/printserver-esp8266 #3: fritzing aggiungere porta parallela

atrent, 2018-01-09:
c'è il componente "connettore centronics" ;)

gianluca-nitti, 2018-01-10:
In Fritzing 0.9.3 scrivendo Centronics ho trovato solo quello a 50 pin per l'SCSI, mentre quello per la parallela/IEEE 1284 lato stampante è il Centronics a 36 pin. Comunque pensavo di mettere il DB25, che c'è in Fritzing ed è quello "lato pc" dei cavi.

Oggi pomeriggio intendo fare delle prove con uno shift register (74HC595), perchè sto usando già 10 GPIO del NodeMCU (8 linee dati + strobe + busy) e quindi se non ne libero almeno 4 non riesco a collegare i pin della parallela che servono per fare la lettura di informazioni sulla stampante. Poi aggiorno il file Fritzing includendo lo shift register ed il connettore DB25.

atrent, 2018-01-10:
hai ragione, è solo SCSI

ok, ottimo

gianluca-nitti, 2018-01-25:
Aggiunto il DB25 già da un po', ma avevo dimenticato di chiudere la issue

closed 2018-01-25 by gianluca-nitti