gianluca-nitti/printserver-esp8266 #6: a prop di USB

atrent, 2018-01-28:
https://hackaday.com/2016/09/03/software-usb-on-the-esp8266/

gianluca-nitti, 2018-01-28:
Interessante, però [sembra solo lato device](https://hackaday.com/2016/09/03/software-usb-on-the-esp8266/#comment-3180150) (emula ad es. tastiere/mouse) mentre qui servirebbe lato host (emulare un pc), e per ora non ho trovato niente per farlo solo in software.
Avevo trovato questo https://github.com/gdsports/esp8266-usb-host-demos che usa dell'hardware aggiuntivo, appena riesco cerco informazioni su quella shield e mi leggo [questo](http://www.usb.org/developers/docs/devclass_docs/usbprint11a021811.pdf) per farmi un'idea di quanto sia fattibile.

atrent, 2018-01-29:
corretto, mi sono entusiasmato troppo presto

gianluca-nitti, 2018-01-30:
Credo di aver trovato l'integrato che fa al caso nostro: http://www.8bit.8u.cz/Files/CH375/CH375DS1.pdf

Può funzionare sia in modalità host che device. Cercando CH375B sui noti siti di e-commerce si trovano delle breakout board complete di porta usb tipo A femmina a circa 10€, a breve ne ordino una per fare prove.

A quanto ho capito da una rapida occhiata al datasheet può funzionare sia via seriale UART che tramite un bus parallelo a 8 bit (potrei riciclare lo stesso usato per la parallela Centronics se si vuole utilizzare una sola stampante - o parallela o USB), comunque in questi giorni approfondisco.

Purtroppo la disponibilità di librerie già pronte per la piattaforma Arduino non sembra granchè, perchè le librerie per USB host sembrano essere quasi tutte destinate ad un altro integrato, il MAX3421E, che ho trovato solo su schede in formato shield per Arduino Uno e simili, che a mio avviso sono scomode da collegare ad un NodeMCU/wemos. Ho trovato questa - link a [header](https://github.com/Makeblock-official/Makeblock-Libraries/blob/master/makeblock/src/MeUSBHost.h) e [implementazione](https://github.com/Makeblock-official/Makeblock-Libraries/blob/master/makeblock/src/MeUSBHost.cpp) -
 che è parte del framework software di uno di quei kit "modulari" per semplificare Arduino, il [MakeBlock](http://www.makeblock.com/) (che ha un [modulo USB host](http://store.makeblock.com/me-usb-host/) basato sul CH375B). Valuterò quanto è scomodo usare quella senza il resto del framework rispetto a scrivere il tutto "da zero" basandomi sul datasheet e su quella libreria (mi sembra di aver capito che tutto il codice sia nel file MeUSBHost.cpp, tolti i commenti e la parte per la gestione di periferiche HID - mouse e tastiere - non sembra neanche troppa roba).

atrent, 2018-02-01:
ottimo lavoro

closed 2018-02-01 by atrent