gianluca-nitti/printserver-esp8266 #7: problemas to connect to de print server

Dr-Gandalf, 2018-07-05:
I am using the only way windows does not complains fo the host name

[image]

but the serial monitors reports this.

_Received IPP request; Version: 0x0100, OperationId: 0x000B, RequestId: 0x0000000B
Unsupported IPP version
Closing connection!_

gianluca-nitti, 2018-07-09:
Thank you for reporting. This is probably due to the fact that the IPP implementation is not fully compliant with the RFC and thus not guaranteed to work with any IPP client - I mostly tested with [CUPS](https://www.cups.org/) on Linux as client, but never on Windows.

I'll see if I can manage to do some testing with Windows and keep this issue updated.

luc-git, 2024-01-31:
It's because windows tells that the ipp version supported is 1.0 and the project supports the version 1.1.
https://github.com/gianluca-nitti/printserver-esp8266/blob/afc98315b0b02709cbbfb3e4fd371374cd6d98c1/printserver/IppStream.h#L26 I just changed `#define IPP_SUPPORTED_VERSION 0x0101 ` by `#define IPP_SUPPORTED_VERSION 0x0100` and the error is gone

luc-git, 2024-01-31:
@Dr-Gandalf