gianluca-nitti/printserver-esp8266 #9: *.fzz files could not be opened

fow0ryl, 2019-06-26:
The fritzing-*.fzz files could not be opened by fritzing 0.9.3 with windows 10 64bit and manjaro-linux 64-bit.
Please provide *.pdf files, so that anyone can view the content.

atrent, 2019-06-27:
they can be opened (I tried now) with 0.9.3 on Debian, I believe it's the same version as manjaro

gianluca-nitti, 2019-06-29:
Here are the exported schematic PDFs:
[fritzing-usb-ch375_schem.pdf](https://github.com/gianluca-nitti/printserver-esp8266/files/3341927/fritzing-usb-ch375_schem.pdf)
[fritzing-parallel_schem.pdf](https://github.com/gianluca-nitti/printserver-esp8266/files/3341928/fritzing-parallel_schem.pdf)

Out of curiosity, which error is Fritzing giving you? I'm also using 0.9.3 and opening the files without any problem.

fow0ryl, 2019-06-30:
Thank you very much for your assistance.
I'm sorry, the "fritzing error" is my fault.
Just installed fritzing 0.9.3 beta on Manjaro and tried to open the fzz files. It gives me an error, that more then 149 parts are missing.
Then I downloaded fritizing and tried it on Windows 10. Opening the fzz file it run's until it get's 49% loaded. Then I get the message "*.fzz could not be opened zip.open(): %d"
Since I never used fritzing before, I don't understand to handle that error...
In the meantime I found that there are errors in the downloaded fzz files (checksum mismatch).
Fritzing is not able to check if files are correct fzz files. It simple tries to open the files and break's.
Since I don't need fritizing for anything and I don't know anybody who is using it (Using eagle for years, changed to KiCad some time ago) I removed fritzing from my system now.
I think standard file format's are the best way for publishing. So anybody can have a look without installing new and unknown software.
Thank you again, again.

fow0ryl, 2019-06-30:
Found that there are some unlabled pin's at the CH375. It would be nice to label them ..

OlehNazar, 2019-10-08:
is it possible to use with this one ?
[image]

gianluca-nitti, 2019-10-08:
@OlehNazar yes, that looks identical to the module I used.

Remember that for USB support to work, the [CH375 library](https://github.com/gianluca-nitti/CH375-Arduino) needs to be installed. Also, as a reminder, I can't guarantee compatibility with any USB printer (especially multi-function ones, i.e. with builtin scanner, as they have more complex USB descriptors which may not be supported by the library at the moment).

closed 2019-10-08 by gianluca-nitti