gianluca-nitti/printserver-esp8266 #11: ESP doesn't power

pavlinb, 2020-04-12:
Hello,

I used direct connection to printer, using 10 GPIO.

ESP8266 doesn't power on, when it is connected to printer HP LaserJet 1100 MS.

gianluca-nitti, 2020-04-15:
Hello,

unfortunately it's quite hard to help debugging an hardware issue with little information, can you provide a diagram? Which ESP module (NodeMCU, Wemos, etc) are you using and how is it powered?

pavlinb, 2020-04-15:
Hi, sorry for missing information.

I use following definition for I/O in NODEMCU:

const int nStrobe =15 ;    //D8 - pin 11 lpt
const int data_0 = 16;    //D0 - pin 2 lpt
const int data_1 = 5;   //D1 - pin 3 lpt
const int data_2 = 4;     //D2 - pin 4 lpt
const int data_3 = 0;     //D3 - pin 5 lpt
const int data_4 = 2;    //D4 - pin 6 lpt
const int data_5 = 14;    //D5 - pin 7 lpt
const int data_6 = 12;    //D6 - pin 8 lpt
const int data_7 = 13;     /D7 - pin 9 lpt
const int busy = 3;       //RXD0 - pin 1 lpt

I connected pins from NODEMCU directly to printer cable.

If I power-on NODEMCU without printer cable connected, ESP starts and  WEB part works. If later  connect the printer cable, printer doesn't want to print.

But if I power-on NODEMCU with printer cable connected, ESP doesn't start at all.

Hope this helps.

Regards.

gianluca-nitti, 2020-04-15:
Here are some suggestions:
* I'd make sure all the unused hardware-related stuff (usb and serial printer) is commented out in `printserver.ino` so that we can be sure there aren't pin conflicts. I suppose you have de-commented lines 32-35 (I'm looking at the [printserver.ino on the master barnch](https://github.com/gianluca-nitti/printserver-esp8266/blob/master/printserver/printserver.ino)), fixed the pin mappings in there according to your wiring, and commented out lines 44-48. I'd also comment out line 50 and make the `printers` array contain just a single element, the reference to `DirectParallelPortPrinter printer` declared on line 35. Also, I'd comment out the call to `Serial.begin` on line 57 which likely conflicts with the fact that pin 3 (serial RX) is used for lpt busy signal. There *should* be no need to comment out the other Serial usages as if I recall correctly they just do nothing if it's not initialized but please take this with a grain of salt because it's been a while since I last looked at the implementation. You might want to look at the serial implementation to see if this is correct or just comment out all the Serial usages as a last resort.
* Also, as an alternative to using RXD0 for busy, which is shared with the ESP hardware serial port, you may want to try other pins such as 9 (SD2) or 10 (SD3).
* Make sure the grounds of the ESP and one of the grounds of the LPT port are tied together. Though in my - limited - understanding of electronics if this is missing this should "just" prevent it from printing and not from powering on.

pavlinb, 2020-04-15:
Thanks for detailed comments.

I have started with pins  9 (SD2) BUSY and  10 (SD3) as STROBE, but the situation was same. Then I found a good disscution here https://www.esp8266.com/viewtopic.php?f=11&t=20727&start=12 and changed mentione 2 pins. 

I then tried very basic example from your link https://forum.arduino.cc/index.php?topic=74776.0, but the printer is still non-responsive.

Is it a problem that NODEMCU is 3.3V , but LPT standard requires 5V?
Should I use shift register in orde to communicate with printer? 

Kind regards.

gianluca-nitti, 2020-04-15:
> Is it a problem that NODEMCU is 3.3V , but LPT standard requires 5V?

Back when I made this project I read somewhere that 3.3V was enough to be read as high, and it in fact worked with the printer I was using, but likely the proper way, as you say, would be using level shifters. Maybe your printer is more picky.

> Should I use shift register in order to communicate with printer?

Unfortunately it shouldn't make any difference in my understanding, apart from reducing how many lines you need to level shift if you run the register on 5V, with voltage shifting between the ESP and the register as noted in the next page of the discussion you linked on the ESP forums.

pavlinb, 2020-04-15:
Thank you for you time.

I'll try some of suggestions.

But seems that my printer is stupid :-)

koller18, 2020-08-14:
> Thank you for you time.
> 
> I'll try some of suggestions.
> 
> But seems that my printer is stupid :-)

Bro, did anything happen? The same printer)

pavlinb, 2020-08-14:
> > Thank you for you time.
> > I'll try some of suggestions.
> > But seems that my printer is stupid :-)
> 
> Bro, did anything happen? The same printer)

Hi, sorry, i left the project.

I nought raspberry pi zero and the problem is solved.

koller18, 2020-08-16:
> > > Thank you for you time.
> > > I'll try some of suggestions.
> > > But seems that my printer is stupid :-)
> > 
> > 
> > Bro, did anything happen? The same printer)
> 
> Hi, sorry, i left the project.
> 
> I bought raspberry pi zero and the problem is solved.

Please give me the connection diagram for the raspberry pi zero. Please, please, please)

pavlinb, 2020-08-16:
I folowed this link:

https://howchoo.com/g/ntvlodqwmzq/wireless-printer-raspberry-pi

Works 100%.

Regards,

koller18, 2020-08-16:
> I folowed this link:
> 
> https://howchoo.com/g/ntvlodqwmzq/wireless-printer-raspberry-pi
> 
> Works 100%.
> 
> Regards,

I am not interested in rasperry pi itself, only the connection to LPT itself. Which LPT pins you have enabled. I understand the middle level)

pavlinb, 2020-08-16:
I used LPT to USB converter.
https://m.aliexpress.com/item/4000175557370.html?pid=808_0000_0131&spm=a2g0n.search-amp.list.4000175557370&aff_trace_key=&aff_platform=msite&m_page_id=8150amp-su3qjz458t79GPRu4GIMbg1597596723333&browser_id=de5a313006a445259038f6db430c8557&is_c=Y