gianluca-nitti/printserver-esp8266 #12: RAW print for ESC/POS printers

ismaelit, 2020-06-19:
Hello! 

Thank you for this great lib! 

I am trying to print on a RS232 ESC/POS Thermal printer, I was successful printing with a RS232<=>TTL adapter (MAX3232 chip) on a dedicated basic program for learning. 

Now I am trying to use printserver-esp8266 to open the printer to the wifi network. 

What I've made: 

Created a new softwareSerial port: 
SoftwareSerial swSerial(D7,D8,false,32);
SerialPortPrinter printer2("SoftwareSerial", &swSerial);

Changed this in this lib: 
Printer* printers[] = {&printer2};

And when I send some lines: 
echo "HELLO WORLD" > /dev/tcp/10.0.2.180/9100
on a remote computer, I receive only the first characters, and nothing more. 

I think this is something related to buffer, because this type of printer just receive RAW HEX commands and plain texts, and not the usual graphical print files. 

Anyone can help on that? Thanks :D

gianluca-nitti, 2020-06-20:
Hello, first of all as a simple tip to help isolationg the problem I'd suggest to make sure there are no `SerialPortPrinter`/`USBPortPrinter`/`DirectParallelPortPrinter`/`ShiftRegParallelPortPrinter` declarations other than your `SerialPortPrinter printer2`, to ensure there are no pin conflicts.

> I receive only the first characters

Sorry for the dumb question but I'm not sure what you mean by *receive* - are some characters actually printed? Or you are getting feedback through some other ways that they have been received by the ESP8266 but nothing is printed?

Maybe you need to send a "flush buffer" command to make it actually print? For example, I remember that with most HP desktop parallel printers you had to send `0x1B 0x45` as shown in the `resetPrinter` function of the code in [this forum post](https://forum.arduino.cc/index.php?topic=74776.0).

ismaelit, 2020-06-22:
Hello Nitti, 

> Hello, first of all as a simple tip to help isolationg the problem I'd suggest to make sure there are no `SerialPortPrinter`/`USBPortPrinter`/`DirectParallelPortPrinter`/`ShiftRegParallelPortPrinter` declarations other than your `SerialPortPrinter printer2`, to ensure there are no pin conflicts.

I believe I have no conflict of ports, because I can print some characters eventually.

> Sorry for the dumb question but I'm not sure what you mean by _receive_ - are some characters actually printed? Or you are getting feedback through some other ways that they have been received by the ESP8266 but nothing is printed?

Sorry, _receive_ means that the data goes to the printer, and the printer receive and print, at first. 

> Maybe you need to send a "flush buffer" command to make it actually print? 

Yes this type of printer needs a initialize command below: 
`swSerial.write('\x1b'); swSerial.write('@'); //initialize`

I am sending this command also, but no progress. See the attached file. 

The first lines before cut, are my initialization script (on `setup()`), that shows the communication working right. 

After the cut, a send via terminal command: 

`echo "HELLO WORLD" > /dev/tcp/10.0.2.180/9100`

And as you see, only the first "HELL" text is printed, even if I send a lot of echoes... 
Sometimes it prints HEL, other times HELLO WORLD (just one line), and does not print anymore. After the reset is pressed on ESP8266, you see the header with SSID etc printed again, and a cut. 

Any ideia? 
[image]

gianluca-nitti, 2020-06-22:
Thanks for the info.

The fact that the initialization message with network info generated by your code gets printed fine but the received data doesn't makes me think the serial communication between the esp8266 and the printer is fine, while something in the code handling the TCP connection is not...

Unfortunately, for now I don't really have any idea at what exactly could be wrong but here are is a tip which may help debugging the issue:
There are some debug prints around the code, you should be able to see their output from the hardware serial of the esp8266 (the one tied to the usb<->serial converter on many boards, I remember I simply used the Arduino IDE serial monitor for this). For example you should be able to see the `Connected` and `Disconnected` from TcpPrintServer.cpp's lines 66 and 39. Maybe you could also add something like `Serial.print("printing char "); Serial.print(b, HEX); Serial.println();` to `SerialPortPrinter::printByte(byte b)` in `SerialPortPrinter.cpp` to have on the main serial a full dump of what's happening and see if the sequence (Connected -> printing chars -> Disconnected) looks correct.