gianluca-nitti/printserver-esp8266 #13: sketch assembly error

Maker39, 2020-06-26:
Greetings!
I need help.
I have WinXP, Arduino IDE 1.8.9, ESP8266 by ESP8266 community version 2.7.1 and Wemos D1 mini Pro.
I am trying to compile the printserver.ino sketch without any changes  (installed the  CH375 and
CH375USBPrinter libraries from the ZIP ).
I get an error:
"fork/exec C:\Documents and Settings\Admin\Local Settings\Application Data\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3.exe: %1 is not a valid Win32 application.
Error compiling for board LOLIN(WEMOS) D1 R2 & mini."
How to fix this error?

And another question: I need to connect the printer as direct parallel. To do this, you will only need to include the appropriate section in the sketch, or will you need to do someone else?

gianluca-nitti, 2020-06-27:
Hello,

the build error doesn't seem related to this particular sketch but rather to the Arduino/ESP8266 IDE setup. There are similar reports, for example, [here](https://github.com/esp8266/Arduino/issues/6470) and [here](https://forum.arduino.cc/index.php?topic=654789.0).

> And another question: I need to connect the printer as direct parallel. To do this, you will only need to include the appropriate section in the sketch, or will you need to do someone else?

There should be no other adjustments required besides editing the main file (`printserver.ino`) to resemble your hardware configuration.

Maker39, 2020-06-28:
thanks for the answer.
I read this and reinstalled ESP8266 to version 2.5.0 - the error is gone.
Now when compiling printserver.ino (without any changes) I have 
**no matching function for call to 'CH375::setBaudRate(int, USBPortPrinter::ensureInitialized()::__lambda0)'**

gianluca-nitti, 2020-08-13:
Sorry for the delay.

The `no matching function for call to 'CH375::setBaudRate(int, USBPortPrinter::ensureInitialized()::__lambda0)'` was caused by the fact that [lambdas that capture arguments cannot be passed as function pointers](https://stackoverflow.com/a/28746827) (it worked a while ago so I assume it worked accidentally in older gcc versions and was removed to comply with the C++ spec).

I fixed the library so that it uses the proper `std::function` type in commit [6a78463](https://github.com/gianluca-nitti/CH375-Arduino/commit/6a7846306f029d99621b5b8d5f409cc859f955e2); with the updated library I can now compile the sketch successfully.

closed 2020-08-13 by gianluca-nitti