gianluca-nitti/printserver-esp8266 #14: Please, help(

koller18, 2020-08-13:
Hello! There is an hp laserjet 1100 printer with lpt. Could you give me a sketch for DirectParallelPortPrinter, but I can't do it. I'm not very good at programming. And tell me what you need to change for esp32. 😘

gianluca-nitti, 2020-08-13:
Hello,

you can use a `DirectParallelPortPrinter` by replacing in `printserver.ino` everything between the last `#include "PrintQueue.h"` and `void setup()` with the following snippet:

```
#define STROBE 10
#define BUSY 9
int DATA[8] = {D0, D1, D2, D3, D4, D5, D6, D7};
DirectParallelPortPrinter printer("lpt1", DATA, STROBE, BUSY);

Printer* printers[] = {&printer1};

#define PRINTER_COUNT (sizeof(printers) / sizeof(printers[0]))
TcpPrintServer server(printers, PRINTER_COUNT);
```

however, depending on the board you are using, you may need to change the following values in the first 3 lines: `10` with the number of the pin connected to the LPT port's strobe pin, `9` with the pin connected to the LPT's busy pin and `D0...D7` with the pins connected to the 8 LPT data pins.

Unfortunately I don't have experience with the ESP32 and never tested this project on one but it should work as long as you have [arduiino-esp32](https://github.com/espressif/arduino-esp32) installed.

Also, make sure to update the (CH375-Arduino)[https://github.com/gianluca-nitti/CH375-Arduino] library to the latest version, otherwise it will likely fail to compile (see final comments on #13 for details).

closed 2020-08-13 by gianluca-nitti

koller18, 2020-08-13:
Please help! I rewrote your firmware for esp32. Almost everything worked, except for mounting the file system. When running in logs writes this: 

**[E][vfs_api.cpp:27] open(): lpt1 does not start with**

Where to dig?