gianluca-nitti/printserver-esp8266 #15: Question 3.3V vs 5V

toncho11, 2020-09-25:
Hi,

You are using only 3.3 volts? The esp8266 works with 3.3V and the parallel port with 5V.

I mean instead of sending 5V on the parallel port you send 3.3V ... because anything above 2.8 volts is considered positive ? 

I do not see any voltage regulator.

toncho11, 2020-09-25:
Ok, so the parallel port can work with 3.3V. Both 5V and 3.3V are possible.

gianluca-nitti, 2020-09-25:
Hi,

yes, the [specification](https://ieeexplore.ieee.org/document/469123) (behind paywall, unfortunately - you can probably obtain it if you have an university or company IEEE subscription) says that the logic high should be at least 2.4v, so 3.3v is fine.

Also, in the other direction (for the BUSY line if I recall correctly) the printer can output up to 5v to signal a logic high.

If you Google it you'll find there is some discussion around on whether GPIOs on ESP8266 boards are 5v-tolerant. Personally, using NodeMCUs I never had problems regarding this but I'd say there is indeed some risk since it looks like there is no official answer.

closed 2020-09-25 by toncho11