gianluca-nitti/printserver-esp8266 #16: CH375-Arduino does not work with EspSoftwareSerial

garik816, 2022-01-26:
can't compile because of the line in the source code:
`SoftwareSerial ch375swSer(CH375_RX, CH375_TX, false, 32);`

> printserver:47:56: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(const uint8_t&, const uint8_t&, bool, int)'
   47 | SoftwareSerial ch375swSer(CH375_RX, CH375_TX, false, 32);
      |                                                        ^
In file included from C:\Users\garik\Downloads\printserver-esp8266-master\printserver\printserver.ino:21:
C:\Users\garik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\espsoftwareserial-main\src/SoftwareSerial.h:93:5: note: candidate: 'SoftwareSerial::SoftwareSerial(int8_t, int8_t, bool)'
   93 |     SoftwareSerial(int8_t rxPin, int8_t txPin = -1, bool invert = false);
      |     ^~~~~~~~~~~~~~
C:\Users\garik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\espsoftwareserial-main\src/SoftwareSerial.h:93:5: note:   candidate expects 3 arguments, 4 provided
C:\Users\garik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\espsoftwareserial-main\src/SoftwareSerial.h:89:5: note: candidate: 'SoftwareSerial::SoftwareSerial()'
   89 |     SoftwareSerial();
      |     ^~~~~~~~~~~~~~
C:\Users\garik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\espsoftwareserial-main\src/SoftwareSerial.h:89:5: note:   candidate expects 0 arguments, 4 provided
Используем библиотеку ESP8266WiFi версии 1.0 из папки: C:\Users\garik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi 
Используем библиотеку espsoftwareserial-main версии 6.15.2 из папки: C:\Users\garik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\espsoftwareserial-main 
Используем библиотеку CH375-Arduino-master версии 0.1.0 из папки: C:\Users\garik\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\CH375-Arduino-master 
exit status 1
no matching function for call to 'SoftwareSerial::SoftwareSerial(const uint8_t&, const uint8_t&, bool, int)'

if I remove "32" everything compiles but doesn't work. the program hangs when accessing the printer (getInfo()). there are no messages in serial. just hangs.

I understand EspSoftwareSerial is no longer compatible with CH375-Arduino.
please help to reconcile them. or choose the versions that are suitable for each other.

Tr0llmops, 2023-04-17:
I'm having this problem too. I've been messing around and trying to fix it - no luck. When I change the initialization, the code compiles and i can even connect to the Wifi as long as no printer is connected, but as soon as I connect the CH375B, everything freezes.

Seems a bit like @gianluca-nitti has given this up? That would really be a shame.
Maybe someone from https://github.com/plerup/espsoftwareserial could point us in the right direction...

My modified init code (printserver.ino):
```C++
    #define CH375_TX D3
    #define CH375_RX D6
    #define CH375_INT D4
--> SoftwareSerial ch375swSer;
    USBPortPrinter printer1("usb", ch375swSer, CH375_INT);
    
    SerialPortPrinter printer2("serial", &Serial);
    Printer* printers[] = {&printer1, &printer2};
    
    #define PRINTER_COUNT (sizeof(printers) / sizeof(printers[0]))
    TcpPrintServer server(printers, PRINTER_COUNT);
    
    void setup() {
      Serial.begin(115200);
-->   ch375swSer.begin(9600, SWSERIAL_8N1, CH375_RX, CH375_TX, false, 32); //TODO: probably remove, should be handled by library or by the USBPortPrinter class
      Serial.println("boot ok");
      SPIFFS.begin();
      for (unsigned int i = 0; i < PRINTER_COUNT; i++) {
        printers[i]->init();
      }
      Serial.println("initialized printers");
      WiFiManager::wifi_setup();
      server.start();
      Serial.println("setup ok");
    }
````

PCSDIAS, 2024-03-09:
See issue #26.

spectrejie, 2024-10-16:
> I'm having this problem too. I've been messing around and trying to fix it - no luck. When I change the initialization, the code compiles and i can even connect to the Wifi as long as no printer is [...]

I also encountered the same problem, not connecting CH375 or even as long as not connecting to D4 pin, wifi can connect successfully.

Have you solved this problem now？