gianluca-nitti/printserver-esp8266 #21: ESP8266 restarts in cycle

InsaneEngineer, 2022-10-05:
I load this programm to ESP8266 D1 mini module and disable USB protocol. It uploads successfully, but ESP don`t boot up and restarts every time. And i have a question: does it work with Samsung ML-4500 printer? It works only for HP printers?

gianluca-nitti, 2022-10-05:
Besides removing the USB printer, did you configure any other printer, e.g. parallel? If yes, you may want to make sure that the pinout doesn't interfere with pins which have special functions (e.g. maybe you are using a pin which is also the RESET pin); the code in `printserver.ino` was tested on a NodeMCU, thus maybe it's using some pin which isn't freely usable on your board.

Also, what do you exactly mean by restarting? Do you see any output on the serial port? 

> And i have a question: does it work with Samsung ML-4500 printer?

It should work as long as you have the proper driver on your computer and can configure it to send the print job over the network (AppSocket/JetDirect or IPP).

gianluca-nitti, 2022-10-05:
I haven't tested it with Windows, only CUPS on Linux. From a quick search, it may be that Windows does not recognize `ipp://` URLs but needs `http://` for the IPP protocol, so you may try to use `http://192.168.1.25:631/lpt1` instead of `ipp://192.168.1.25:631/lpt1`. So I'd try this as a first attempt.

If this still doesn't work, consider that the IPP protocol is quite complex and its server implementation in this project is not 100% compliant to the RFC spec (just enough to make it work with CUPS), thus it may not work well together with the Windows client implementation. Thus, you may want to try using the AppSocket/JetDirect protocol which is way simpler (= less risk of incompatibilities between server and client implementation).

To do so, you would need to modify the source code in the `.ino` file by removing the useless serial printer (this is needed because only one printer can be exposed via this protocol), by removing the line `SerialPortPrinter printer2("serial", &Serial);` and changing the line `Printer* printers[] = {&printer1, &printer2};` with `Printer* printers[] = {&printer1};` or whatever you called your instance of `DirectParallelPortPrinter`/`ShiftRegParallelPortPrinter` if it's not `printer1`.

After doing this, from the web interface you should see only your `lpt1` printer, accessible via `socket://192.168.1.25:9100` as well. It looks like however that Windows wouldn't accept this URL format as well, and you'll likely need to follow a procedure [like this](https://dfarq.homeip.net/add-a-tcpip-printer-in-windows-10/) (look at the screenshots) and put `192.168.1.25` and `9100` in the two separate `Hostname or IP Address` and `Port name` fields respectively.

InsaneEngineer, 2022-10-05:
I connected to ESP via TCP/IP connection. I just copied 192.168.1.25:9100 and it works. Thx. In future i`m solder a cable for LPT port from old broken USB LPT adapter.

gianluca-nitti, 2022-10-05:
Good to hear! So you maneged to print succesfully? Out of curiosity, how did you connect the printer, directly to the D1 mini or via a shift register?

closed 2022-10-05 by gianluca-nitti

InsaneEngineer, 2022-10-05:
No. I`m connected directly - and ESP8266 shuts down memntaly.