gianluca-nitti/printserver-esp8266 #22: Missing labels and connections on fritzing-usb-ch375.fzz

bobbygituser, 2022-12-29:
On the fritzing file for the USB connection using CH375, "fritzing-usb-ch375.fzz", some of the pins on the CH375 do not have labels. There is also no indication on the connection for the pins between the USB socket and the CH375. Please help to clarify these.

[image]

gianluca-nitti, 2023-01-03:
The reason is that the CH375 part in that schematic is not a bare CH375 chip but a breakout board like the one picured below, with built-in USB socket. Thus the pins on the Fritzing part are just the exposed pins on the breakout board.

The unconnected ones are not needed. The chip supports both a serial and a parallel communication mode, and the [library](https://github.com/gianluca-nitti/CH375-Arduino) used in the sketch only uses the serial mode, thus the parallel data and control lines are left unconnected.

[image]

closed 2023-01-03 by gianluca-nitti

bobbygituser, 2023-01-04:
@gianluca-nitti thanks for the clarification. not sure if this is nitpicking, but would you kindly reflect that info in the README section? The breakout board / module wasn't apparent in the README nor in the fritzing. It would help new readers and probably deter similar questions to mine.

I would suggest to rename the part as CH375(B) module, or something similar to that.

gianluca-nitti, 2023-01-09:
Added a comment in the readme. Thanks for reporting.

Augustoklippel, 2023-06-03:
Hi, i has buy a module CH375B equal published image and not working.
Can you identify what is pins in fritzing in module?