gianluca-nitti/printserver-esp8266 #24: Print client for ipp

deepakkoirala, 2023-03-30:
I know this library is for creating a print server.

How about creating a print client from esp32/8266. I want to make a print request from esp itself to a network printer using the native protocol.

gianluca-nitti, 2023-03-31:
Unless you can find a pre-made library which is small andl flexible enough to be adaptable to such an embedded environment (there is none I know of), you need to implement a subset of IPP protocol suitable to your purpose.

It should not be particularly difficult as IPP is actually HTTP-based. Maybe you can start from [here](https://www.pwg.org/ipp/ippguide.html) to find some useful information.

closed 2023-03-31 by gianluca-nitti

deepakkoirala, 2023-04-02:
Any plans to implement as part of this library. I did not find any. This would be first and really helped everyone.

I have a inkjet printer, trying to schedule print from esp. Since esp is low power.

gianluca-nitti, 2023-04-03:
I kinda understand your use case but really I don't see it as being related to this project, as a client and a server serve very different purposes.

It shouldn't be much work to implement a basic client using ESP8266HTTPClient or similar libraries and the IPP RFCs as documentation, however I'm not going to do it (no time nor much interest).

deepakkoirala, 2023-04-07:
Yeah correct but need a easy implementation. Thats it.