gianluca-nitti/printserver-esp8266 #26: Finally I was able to compile it

PCSDIAS, 2024-03-09:
Had to make the following changes in printserver.ino in order to compile it. 
Pin defines  CH375_TX, CH375_RX D6 and CH375_INT are valid if Board "NodeMCU 1.0 " is selected.
Altough compiled don't know if it is going to work in real life.

1. changed line 47 from:
    `SoftwareSerial ch375swSer(CH375_RX, CH375_TX, false, 32)`;
   to: 
    `EspSoftwareSerial::UART ch375swSer;` 

2. changed line  58 from:
    `ch375swSer.begin(9600);`
   to:
    `ch375swSer.begin(BAUD_RATE, EspSoftwareSerial::SWSERIAL_8N1, CH375_RX, CH375_TX, false, 32, 11);`

3. added a line after line 46
    `#define BAUD_RATE 9600`

vodibe, 2024-05-16:
Good to know, thanks for the tips. I followed all (except n. 4) and indeed, the code compiles correctly on my NodeMCU v3. However, in my case, i'm not able to turn on the NodeMCU by following the wiring schema [fritzing-usb-ch375.fzz](https://github.com/gianluca-nitti/printserver-esp8266/blob/afc98315b0b02709cbbfb3e4fd371374cd6d98c1/fritzing-usb-ch375.fzz)
In my case, the printer (Samsung M2070 with a USB 2.0 port 0.5 A) is connected to the ch375. Did you try the wiring irl?

spectrejie, 2024-10-16:
Thank you for your tips.Finally able to compile successfully.However, it could not be successfully launched in ESP8266.
My printer is an HP LazerJet Pro MFP M126a.
One problem is that when you connect the printer's USB port, ESP8266 is not lit. My understanding is that the printer should be able to power the ESP8266, I don't know if my understanding is wrong.
Another problem is that even if there is power, ESP8266 does not start properly.
When I don't connect the printer, and I don't connect the CH375 module (or even just keep PIN D4 not connected), the ESP8266 can boot and can connect to wifi.
However, as long as the D4 pin is connected, the serial port output is garbled. Not even "boot ok" is output.